matthias-bs/BresserWeatherSensorTTN #13: Error compiled

Lordcybertracker, 2022-10-27:
Guten morgen,
ich hab das ganze in PlatformIO importiert.
Allerdings bekomme ich:
In file included from C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:142:0:
lib/BresserWeatherSensorReceiver-main/src/WeatherSensor.h:196:25: error: 'NUM_SENSORS' was not declared in this scope
         sensor_t sensor[NUM_SENSORS];      //!< sensor data array
                         ^
lib/BresserWeatherSensorReceiver-main/src/WeatherSensor.h: In member function 'void WeatherSensor::clearSlots(uint8_t)':
lib/BresserWeatherSensorReceiver-main/src/WeatherSensor.h:218:30: error: 'NUM_SENSORS' was not declared in this scope
             for (int i=0; i< NUM_SENSORS; i++) {
                              ^
lib/BresserWeatherSensorReceiver-main/src/WeatherSensor.h:219:40: error: 'sensor' was not declared in this scope
                 if ((type == 0xFF) || (sensor[i].s_type == type)) {
                                        ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino: In member function 'void cSensor::setup(uint32_t)':
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:931:44: error: 'class WeatherSensor' has no member named 'sensor'
             if ((ws > -1) && weatherSensor.sensor[ws].valid && weatherSensor.sensor[ws].rain_ok) {
                                            ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:931:78: error: 'class WeatherSensor' has no member named 'sensor'
             if ((ws > -1) && weatherSensor.sensor[ws].valid && weatherSensor.sensor[ws].rain_ok) {
                                                                              ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:932:58: error: 'class WeatherSensor' has no member named 'sensor'
                 rainGauge.update(timeinfo, weatherSensor.sensor[ws].rain_mm, rg_overflow);
                                                          ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino: In member function 'void cSensor::doUplink()':
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1102:70: error: 'class WeatherSensor' has no member named 'sensor'
       DEBUG_PRINTF("Air Temperature:    % 3.1f °C\n",   weatherSensor.sensor[ws].temp_c);
                                                                      ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1103:69: error: 'class WeatherSensor' has no member named 'sensor'
       DEBUG_PRINTF("Humidity:            %2d   %%\n",   weatherSensor.sensor[ws].humidity);
                                                                     ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1104:66: error: 'class WeatherSensor' has no member named 'sensor'
       DEBUG_PRINTF("Rain Gauge:       %7.1f mm\n",      weatherSensor.sensor[ws].rain_mm);
                                                                  ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1105:71: error: 'class WeatherSensor' has no member named 'sensor'
       DEBUG_PRINTF("Wind Speed (avg.):    %3.1f m/s\n", weatherSensor.sensor[ws].wind_avg_meter_sec_fp1/10.0);
                                                                       ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1106:71: error: 'class WeatherSensor' has no member named 'sensor'
       DEBUG_PRINTF("Wind Speed (max.):    %3.1f m/s\n", weatherSensor.sensor[ws].wind_gust_meter_sec_fp1/10.0);
                                                                       ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1107:68: error: 'class WeatherSensor' has no member named 'sensor'
       DEBUG_PRINTF("Wind Direction:     %4.1f °\n",     weatherSensor.sensor[ws].wind_direction_deg_fp1/10.0);
                                                                    ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1115:72: error: 'class WeatherSensor' has no member named 'sensor'
         DEBUG_PRINTF("Soil Temperature 1: % 3.1f °C\n",  weatherSensor.sensor[s1].temp_c);
                                                                        ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1116:71: error: 'class WeatherSensor' has no member named 'sensor'
         DEBUG_PRINTF("Soil Moisture 1:     %2d   %%\n",  weatherSensor.sensor[s1].moisture);      
                                                                       ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:173:51: note: in definition of macro 'DEBUG_PRINTF'
     #define DEBUG_PRINTF(...) { DEBUG_PORT.printf(__VA_ARGS__); }
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1170:51: error: 'class WeatherSensor' has no member named 'sensor'
                         (s1 > -1) ? weatherSensor.sensor[s1].valid : false,
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1171:51: error: 'class WeatherSensor' has no member named 'sensor'
                         (s1 > -1) ? weatherSensor.sensor[s1].battery_ok : false,
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1172:51: error: 'class WeatherSensor' has no member named 'sensor'
                         (ws > -1) ? weatherSensor.sensor[ws].valid : false,
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1173:51: error: 'class WeatherSensor' has no member named 'sensor'
                         (ws > -1) ? weatherSensor.sensor[ws].battery_ok : false);
                                                   ^
C:/Users/laval/Documents/PlatformIO/Projects/Bresser_TTN/src/BresserWeatherSensorTTN.ino:1178:48: error: 'class WeatherSensor' has no member named 'sensor'
         encoder.writeTemperature(weatherSensor.sensor[ws].temp_c);
         
Was habe ich vergessen einzustellen?
Danke

matthias-bs, 2022-10-27:
Hallo,

es scheint Du hast für die Library [BresserWeatherSensorReceiver](https://github.com/matthias-bs/BresserWeatherSensorReceiver) vom main-Branch die LATEST-Version verwendet. Hier wird gerade einiges geändert, was anschließend (nach der nächsten Release) in [BresserWeatherSensorTTN](https://github.com/matthias-bs/BresserWeatherSensorTTN) nachgezogen werden muss.

Zunächst einmal empfehle ich die Versionen aller Libraries laut [package.json](https://github.com/matthias-bs/BresserWeatherSensorTTN/blob/main/package.json) zu verwenden. Eventuell hilft auch das YML-Skript [CI.yml](https://github.com/matthias-bs/BresserWeatherSensorTTN/blob/main/.github/workflows/CI.yml) (automatisches Kompilieren nach Check-In) weiter.

Zu beachten ist, dass ggf. noch zwei Anpassungen nötig sind:
1. Kopieren der `WeatherSensorCfg.h` in den BresserWeatherSensorReceiver-Library-Pfad; in CI.yml:
```
cp /home/runner/work/BresserWeatherSensorTTN/BresserWeatherSensorTTN/WeatherSensorCfg.h /home/runner/Arduino/libraries/BresserWeatherSensorReceiver/src/
```

2. Patchen MCCI_LoRaWAN_LMIC_library der Library-Config; in CI.yml:
```
echo "#define LMIC_ENABLE_DeviceTimeReq 1" >> /home/runner/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/project_config/lmic_project_config.h

# Fix for linker error with Arduino core for ESP32 v2.0.x as suggested in https://github.com/mcci-catena/arduino-lmic/issues/714#issuecomment-822051171
echo "#define hal_init LMICHAL_init" >> /home/runner/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/project_config/lmic_project_config.h
```
Also einfach Einfügen der beiden folgenden Zeile in `Arduino/libraries/MCCI_LoRaWAN_LMIC_library/project_config/lmic_project_config.h`:
```
#define LMIC_ENABLE_DeviceTimeReq 1
#define hal_init LMICHAL_init
```

Zu PlatformIO kann ich leider nicht helfen, da ich es bisher nicht genutzt habe.

Viele Grüße
Matthias

Lordcybertracker, 2022-10-28:
Schade irgendwie bin ich zu blöde dazu. Ich versuch es die Tage nochmal.
Das würde uns viel Arbeit ersparen es so zu lösen. Da wir die Bresser nicht umbauen müssen.
Alternativ hatten wir vor eine neue Platine für die Bresser zu bauen und diese zu tauschen.

matthias-bs, 2022-11-08:
Ich habe es nicht vergessen, ich komme nur gerade nicht dazu. In 1-2 Wochen wird es eine neue Release geben die hoffentlich ohne Probleme zu kompilieren sein wird.

Lordcybertracker, 2022-11-09:
Danke.
ideal wäre dazu dann ein Lib Pakage das man genau die richtigen hat.
Ich hab mich mal mit der Mqtt Variante beschäftigt. Aber auch die bekomme ich nicht zum laufen.
Dein Projekt würde uns aber das neu entwickeln der Platine und ersetzen ersparen.

matthias-bs, 2022-11-10:
Bitte probiere einmal die neue [Release v0.3.1](https://github.com/matthias-bs/BresserWeatherSensorTTN/releases/tag/v0.3.1) mit allen dazugehörenden [aktuellen Libraries](https://github.com/matthias-bs/BresserWeatherSensorTTN/blob/main/package.json). Eine [Anleitung zum Compilieren und Anpassen](https://github.com/matthias-bs/BresserWeatherSensorTTN#software-build-setup) habe ich im [README](https://github.com/matthias-bs/BresserWeatherSensorTTN/blob/main/README.md) hinzugefügt. Die Release compiliert auf der GitHub Platform (CI, siehe [actions/runs/3434467352](https://github.com/matthias-bs/BresserWeatherSensorTTN/actions/runs/3434467352)) und mit meiner lokalen Arduino-Installation fehlerfrei und läuft gerade auf der Zielhardware.

Lordcybertracker, 2022-11-15:
Danke. Ich schau es mir am Wochenende an.

Lordcybertracker, 2022-12-13:
Leider bin ich immer noch nicht dazu gekommen.
Das wird wohl in den Urlaub zwischen den Jahren fallen.
Getestet wird auf einem T-Beam und Heltec Wirless Stick Lite sowie meinem eigenen Board mit ESP32 Pico. 
Davon hab ich noch einige hier.

Die MQTT Variante wäre für Local hier auch Interessant die ans laufen zu bekommen.

Parallel baue ich an einer Ersatzplatine welche durch die in der Bresser befindlichen ersetzt wird. Mal sehen wie das wird.

closed 2023-03-05 by matthias-bs