matthias-bs/BresserWeatherSensorTTN #40: Improve ADC input handling for TTGO LoRa32 V1

matthias-bs, 2023-01-22:
Actually VBAT is provided to the ADC and should be used as uplink data and for checking the supply voltage. 
There is no separate supply voltage measurement by default.

matthias-bs, 2023-01-22:
In https://github.com/LilyGO/TTGO-LORA32/blob/master/schematic1in6.pdf on sheet 4, there are two input dividers - R42/R43 and R44/R45 for VBAT connecting to IO35. Chances are that on of them has a wrong label and is actually connected to VBUS.

matthias-bs, 2023-01-22:
@mczakk Is it possible to connect a power supply to the battery connector and check if you can get a plausible voltage reading from IO34?

matthias-bs, 2023-01-22:
Opened issue https://github.com/LilyGO/TTGO-LORA32/issues/35

matthias-bs, 2023-01-25:
If IO34 is not connected:
Please try this circuit and report if the supply voltage (with and without connected battery ) can be measured

```
                    100k       100k
JP1 Pin2 - GND  o--/\/\/\-–|--/\/\/\------o +5V - JP1 - Pin 1
                           |
                           `--------------> IO34 (ADC in) - JP2 Pin 10 
```

matthias-bs, 2023-01-25:
@franki29 Maybe you can help to figure this out?

mczakk, 2023-01-25:
> @mczakk Is it possible to connect a power supply to the battery connector and check if you can get a plausible voltage reading from IO34?

I'll be honest, I have no idea! Don't even know where the battery connector is. Or how I would get a power supply to it 😱

matthias-bs, 2023-01-26:
@franki29 Thanks for the quick answer, but my question was targeted to the TTGO LoRa32 specific power supply and ADC input circuit design.

In general, I know how to use the ADCs and how to get calibrated conversions.

franki29, 2023-01-27:
Oh sorry, I did not read the full issue, sorry. I have a board like this one https://polluxlabs.net/arduino-tutorials/daten-per-lora-senden-und-empfangen-mit-dem-ttgo-lora32-board/ and I think it is a clone. There are no resistors between the Battery connector and the ADC, none of them. (Measured by a Ohm Meter) So basically to add two 100K like you proposed in this issue should work.