matthias-bs/BresserWeatherSensorTTN #42: Outputting Bresser data on TTGO ESP32 OLED board

mczakk, 2023-02-03:
Hi Matthias, 
I have used your code, and also that of @Uspizig to output the bresser data on the screen of the ESP32 board.
I have put the code in my own repository here: [https://github.com/mczakk/TTGO-ESP32-Bresser-with-display](url)
I am struggling to extract the current time as generated by the function 'printDateTime' (line 975)so that I can print it on the screen (line 1533)
Could you possibly advise?

Many Thanks

mczakk, 2023-02-03:
I've now added updated code, which will produce a string called 'global_tbuf'
which i can print out, (line 1534)
Do you know how i can print this on the oled?

mczakk, 2023-02-03:
I seem to have got it working properly!!, check it out and see what you think?

closed 2023-02-03 by mczakk

matthias-bs, 2023-02-05:
I'll have a look at it later.

matthias-bs, 2023-02-05:
If you create a fork of my repository, you can get updates from it later on, even if you made your own modifications. And the CI workflow (continuous integration) will compile your code whenever you check in any changes and help to check it. (see the item "Actions" at the top)

mczakk, 2023-02-05:
Forgive my ignorance about all things github, but would a fork be like an amended version of your code? And does it matter that I've just mashed up your code with uspizig's? I will happily give you both the credit you deserve!

matthias-bs, 2023-02-06:
No worries!
That's the power of open source and community driven development! The selected MIT license allows and encourages to do this! Did you see how many libraries I'm using in my project and that it is mainly based on the LMIC libraries? So the project is in big parts the result of other people's work with a few changes and additions by myself...

And GitHub makes this sharing, reusing, remixing etc. quite easy and comfortable.

If the modifications of your fork are of general interest, you can create a pull request and I could re-integrate them in my project.

matthias-bs, 2023-02-06:
BTW: Did you see that @Uspizig used some parts of my code in https://github.com/Uspizig/Bresser_Weather_NEW/tree/main/WeatherDecoder_LORA_RFM95Only? ;-)

matthias-bs, 2023-02-06:
I have made a fork and did some modifications (see history for explanation):
https://github.com/matthias-bs/TTGO-ESP32-Bresser-with-display
Your code was valid, but the handling of the time string was unnecessarily complicated. I know, string handling in C is more difficult than it should be...

I hope I did not make any mistakes. Please let me know if it works.

matthias-bs, 2023-02-06:
You have put a delay of 10 and 15 seconds in the display function to allow reading before the contents are changed. This might cause problems in the LoRaWAN protocol handling. After sending the uplink, other things might have to be done in time, e.g. receiving an acknowledgement, parameter changes, the network time etc.

I will try to address this later.

mczakk, 2023-02-07:
Thankyou for sorting out the time string, it took me a while to work that one out, yes string handling does seem overly complex when compared to python!
Your code compiles perfectly, I've made a couple of changes in the display which i will try and change on the fork, hopefully I'll get it right!

Regarding the delays, and potential problems with the timing, is that not just a function of when the display function is run? Would it not be easier to wait until the LoRaWAN bits were done before running the display update, after all, we have five minutes after the data is sent to ttn. 
Is it as simple as just moving the call to 'displayWeatherData()' ?

matthias-bs, 2023-02-07:
Yes, in this case it would be the easiest solution to call `displayWeatherData()` just before entering deep-sleep mode.

The more complicated solution would be to keep the state (current page) inside the function and compare the current time to the last time the function was called and call it again when due. This would be the desired solution if the sketch were to run continuously (instead of going into deep-sleep mode after each cycle).

mczakk, 2023-02-07:
where would be the best place to put it in your opinion?
Also your comment about the [ws] value was interesting, I've not really had any problems with that other than the occasional zero value being reported.
Apart from when i tried to connect the esp32 to my raspberry pi, it seemed to throw up errors in the serial monitor that i wasn't seeing when it was connected to my windows laptop!

matthias-bs, 2023-02-10:
The parity errors above mean that radio signal is disturbed, either directly or by the power supply of your TTGO board. If you want to save battery power in the field, you should switch of these debug messages before deploying the sketch onto the target device.

mczakk, 2023-02-13:
> 

It does throw a compilation error:
```
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:173:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:88:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
                                                                                 ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:173:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino: In function 'void loop()':
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:673:33: error: 'class cSensor' has no member named 'wsIndex'
     displayWeatherData(mySensor.wsIndex);
                                 ^~~~~~~
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino: In member function 'void cSensor::doUplink()':
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:1494:3: error: 'wsIndex' was not declared in this scope
   wsIndex = ws;
   ^~~~~~~
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:1494:3: note: suggested alternative: 'rindex'
   wsIndex = ws;
   ^~~~~~~
   rindex

exit status 1

Compilation error: 'class cSensor' has no member named 'wsIndex'
```
Also, how do i get my repository to update with your chenges?

mczakk, 2023-02-13:
> Regarding the `displayWeatherData()` function call: [matthias-bs/TTGO-ESP32-Bresser-with-display@0390b11](https://github.com/matthias-bs/TTGO-ESP32-Bresser-with-display/commit/0390b11efcc1b76db3a77867a5404ac1a82e4b34) I have moved it to a location just before going to sleep after a successful transmission if nothing else is left to do. I hope I got the passing of the index into the data structure right without compiling/testing...

It does throw a compilation error:
```In file included from /home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:173:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:88:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V1 defined; using on-board transceiver")
                                                                                 ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:173:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->14, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino: In function 'void loop()':
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:673:33: error: 'class cSensor' has no member named 'wsIndex'
     displayWeatherData(mySensor.wsIndex);
                                 ^~~~~~~
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino: In member function 'void cSensor::doUplink()':
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:1494:3: error: 'wsIndex' was not declared in this scope
   wsIndex = ws;
   ^~~~~~~
/home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:1494:3: note: suggested alternative: 'rindex'
   wsIndex = ws;
   ^~~~~~~
   rindex

exit status 1

Compilation error: 'class cSensor' has no member named 'wsIndex'
```

matthias-bs, 2023-02-13:
Sorry for this! Fixed in https://github.com/matthias-bs/TTGO-ESP32-Bresser-with-display/commit/5f11b900be6513aed637ba1ba5bb25a47feac63d

I created a pull request, so you can see the changes and accept or reject them.

mczakk, 2023-02-13:
Hmmm, that seems to have made it worse!
disabled debug level in BresserWeatherSensorCfg.h (line 64)
codde compiled and uploaded ok, but serial monitor dis;lays this:
```
ELF file SHA256: 78cd42da231a6198

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Display should work
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3795  PS      : 0x00060630  A0      : 0x800d383c  A1      : 0x3ffb25d0  
A2      : 0x3ffb265c  A3      : 0x00847628  A4      : 0x00000084  A5      : 0x3ffc590c  
A6      : 0x00008476  A7      : 0x3ffb263c  A8      : 0x3ffc590c  A9      : 0x33be27a0  
A10     : 0x3ffb263c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc590c  
A14     : 0x3ffb268c  A15     : 0x63ffffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00847658  LBEG    : 0x40087150  LEND    : 0x4008715b  LCOUNT  : 0x00000000  

Backtrace: 0x400d3792:0x3ffb25d0 0x400d3839:0x3ffb2600 0x400d339b:0x3ffb2700 0x400d3434:0x3ffb2720 0x400d3474:0x3ffb2740 0x400d43a1:0x3ffb2760 0x400d5809:0x3ffb2780 0x400d5d46:0x3ffb27a0 0x400d6bb1:0x3ffb27c0 0x400d3673:0x3ffb27e0 0x400d30dd:0x3ffb2800 0x400ddda9:0x3ffb2820

ELF file SHA256: 78cd42da231a6198

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
```
and the OLED displays this:
Thanksfor creating the pull request, though i did accept your changes before compiling and uploading!!
[image]

matthias-bs, 2023-02-13:
Nice matrix pattern! ;-)

You could install the exception decoder to locate the error: https://github.com/me-no-dev/EspExceptionDecoder

It's a little bit difficult for me to help you, because I do not have this board. I could modify the display function in order to check if the data is passed to it correctly.  But currently I cannot make any promises when I will find the time.

At least GitHub's versioning will allow you to revert these changes.

Sorry for this!

matthias-bs, 2023-02-13:
I did not notice this compiler warning before:

```
/home/mp/pCloudDrive/BresserWeatherSensor/10-BresserWeatherSensorTTN/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:147: warning: "OLED_SDA" redefined
 #define OLED_SDA 21
 
In file included from /home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/esp32-hal-gpio.h:29,
                 from /home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/esp32-hal.h:79,
                 from /home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/Arduino.h:36,
                 from sketch/BresserWeatherSensorTTN2.ino.cpp:1:
/home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/variants/ttgo-lora32-v1/pins_arduino.h:15: note: this is the location of the previous definition
 #define OLED_SDA 4
 
/home/mp/pCloudDrive/BresserWeatherSensor/10-BresserWeatherSensorTTN/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:148: warning: "OLED_SCL" redefined
 #define OLED_SCL 22
 
In file included from /home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/esp32-hal-gpio.h:29,
                 from /home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/esp32-hal.h:79,
                 from /home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/Arduino.h:36,
                 from sketch/BresserWeatherSensorTTN2.ino.cpp:1:
/home/mp/.arduino15/packages/esp32/hardware/esp32/2.0.6/variants/ttgo-lora32-v1/pins_arduino.h:16: note: this is the location of the previous definition
 #define OLED_SCL 15
```
So... It seems you have selected "Board Revision: TTGO LoRa32 V1 (No TFCard)" in the Arduino IDE, while it should actually be V2 or V21new (I don't remember which).

mczakk, 2023-02-13:
d'oh, it seems i didn't install the right board manager when i moved to using the arduino ide on my linux machine from my windows one, will try your revisions again and report back

mczakk, 2023-02-13:
has WeatherSensorCfg.h been changed at all, as i wasn't getting any of these pragma messages before?
I have tried board revision V2 and V2.1 1.6.1 (the latter is what is printed on my board)
with either i now get nothing on the screen , and no ttn connection

mczakk, 2023-02-13:
trying again with the latest version of BresserWeatherSensorReciever, board set to TTGO esp32 V2.1.1.6.1
ok, got the serial messages back, it's recieving, but bad data, and doesn't seem to be trying to connect at all to ttn
output on compile:
```  
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:62:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:97:85: note: #pragma message: ARDUINO_TTGO_LoRa32_V21new defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V21new defined; using on-board transceiver")
                                                                                     ^
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:62:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->23, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
Sketch uses 634501 bytes (48%) of program storage space. Maximum is 1310720 bytes.
Global variables use 33264 bytes (11%) of dynamic memory, leaving 261648 bytes for local variables. Maximum is 294912 bytes.
esptool.py v4.2.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:e8:ba:54
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000abfff...
Compressed 17440 bytes to 12108...
Writing at 0x00001000... (100 %)
Wrote 17440 bytes (12108 compressed) at 0x00001000 in 1.3 seconds (effective 104.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 382.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 616.2 kbit/s)...
Hash of data verified.
Compressed 635472 bytes to 324355...
Writing at 0x00010000... (5 %)
Writing at 0x0001b004... (10 %)
Writing at 0x0002c808... (15 %)
Writing at 0x0003f025... (20 %)
Writing at 0x00044989... (25 %)
Writing at 0x0004973b... (30 %)
Writing at 0x0004ed51... (35 %)
Writing at 0x000546f6... (40 %)
Writing at 0x00059aed... (45 %)
Writing at 0x0005f107... (50 %)
Writing at 0x00064b54... (55 %)
Writing at 0x0006c492... (60 %)
Writing at 0x00074f2a... (65 %)
Writing at 0x0007d42c... (70 %)
Writing at 0x00082eae... (75 %)
Writing at 0x0008ba4e... (80 %)
Writing at 0x00092951... (85 %)
Writing at 0x0009be54... (90 %)
Writing at 0x000a12c4... (95 %)
Writing at 0x000a67a1... (100 %)
Wrote 635472 bytes (324355 compressed) at 0x00010000 in 28.7 seconds (effective 177.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[... 25 lines cut ...]
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
[    17][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   530][D][BresserWeatherSensorTTN2.ino:598] setup(): Preferences: weathersensor_timeout: 180 s
[   531][D][BresserWeatherSensorTTN2.ino:600] setup(): Preferences: sleep_interval:        300 s
[   536][D][BresserWeatherSensorTTN2.ino:602] setup(): Preferences: sleep_interval_long:   900 s
[   545][D][BresserWeatherSensorTTN2.ino:607] setup(): 
[   550][D][BresserWeatherSensorTTN2.ino:992] printDateTime(): 00:08:40 01-01-70
[   557][D][BresserWeatherSensorTTN2.ino:615] setup(): RTC sync required
[   563][D][BresserWeatherSensorTTN2.ino:621] setup(): myEventlog.setup() - done
[   573][D][BresserWeatherSensorTTN2.ino:1193] getVoltage(): Voltage = 284mV
[   577][D][WeatherSensor.cpp:82] begin(): [SX1276] Initializing ... 
[   595][D][WeatherSensor.cpp:95] begin(): success!
[   596][D][WeatherSensor.cpp:130] begin(): [SX1276] Setup complete - awaiting incoming messages...
[  9021][D][WeatherSensor.cpp:239] getMessage(): [SX1276] Receive failed: [0]
[ 21881][D][WeatherSensor.cpp:239] getMessage(): [SX1276] Receive failed: [0]
[ 22831][D][WeatherSensor.cpp:220] getMessage(): [SX1276] R [D4] RSSI: -76.5
[ 22832][D][WeatherSensor.cpp:756] decodeBresser6In1Payload(): Temp: 0  Hum: 0  UV: 0  Wind: 1  Rain: 1  Moist: 0
[ 33009][D][WeatherSensor.cpp:239] getMessage(): [SX1276] Receive failed: [0]
```

mczakk, 2023-02-13:
Now with board revision set to V2:
compile output:
```
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:173:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:92:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver")
                                                                                 ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:93:66: note: #pragma message: LoRa DIO1 must be wired to GPIO33 manually!
     #pragma message("LoRa DIO1 must be wired to GPIO33 manually!")
                                                                  ^
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorTTN2/BresserWeatherSensorTTN2.ino:173:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->12, CS->14, GD0/G0/IRQ->4, GDO2/G1/GPIO->16
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
/home/matt/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
/home/matt/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
                                                                        ^
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:62:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:92:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver")
                                                                                 ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:93:66: note: #pragma message: LoRa DIO1 must be wired to GPIO33 manually!
     #pragma message("LoRa DIO1 must be wired to GPIO33 manually!")
                                                                  ^
In file included from /home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensor.cpp:62:
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
/home/matt/Arduino/libraries/BresserWeatherSensorReceiver/src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->12, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
Sketch uses 634337 bytes (48%) of program storage space. Maximum is 1310720 bytes.
Global variables use 33256 bytes (11%) of dynamic memory, leaving 261656 bytes for local variables. Maximum is 294912 bytes.
esptool.py v4.2.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:e8:ba:54
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000abfff...
Compressed 17440 bytes to 12108...
Writing at 0x00001000... (100 %)
Wrote 17440 bytes (12108 compressed) at 0x00001000 in 1.3 seconds (effective 105.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 383.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 617.3 kbit/s)...
Hash of data verified.
Compressed 635312 bytes to 324238...
Writing at 0x00010000... (5 %)
Writing at 0x0001b033... (10 %)
Writing at 0x0002c808... (15 %)
Writing at 0x0003efd3... (20 %)
Writing at 0x0004493f... (25 %)
[... 90 lines cut ...]
// This define is set by selecting "Board: TTGO LoRa32-OLED" / "Board Revision: TTGO LoRa32 V1 (No TFCard)"
// in the Arduino IDE:
//#define ARDUINO_TTGO_LoRa32_V1

// This define is set by selecting "Board: TTGO LoRa32-OLED" / "Board Revision: TTGO LoRa32 V2"
[... cut ...]
// Adafruit Feather ESP32S2 with RFM95W "FeatherWing" ADA3232
// https://github.com/espressif/arduino-esp32/blob/master/variants/adafruit_feather_esp32s2/pins_arduino.h
//
// This define is set by selecting "Adafruit Feather ESP32-S2" in the Arduino IDE:
//#define ARDUINO_ADAFRUIT_FEATHER_ESP32S2

// DFRobot Firebeetle32
// https://github.com/espressif/arduino-esp32/tree/master/variants/firebeetle32/pins_arduino.h
//
// This define (not very specific...) is set by selecting "FireBeetle-ESP32" in the Arduino IDE:
//#define ARDUINO_ESP32_DEV

```

matthias-bs, 2023-02-13:
Yes, the board definitions should be left commented out if configuration via the IDE is supported for your board (please see the Readme.md).

And no, you do not get bad data - what you see are the flags which data is available in the current message (well, this is not clear from the output).

Do you get a TTN connection with any of the two board versions? Which pin numbers did work four you previously? Maybe I made a mistake with the pinning.

matthias-bs, 2023-02-13:
For V2, a connection between the ESP and the LoRa radio (which is only used by LMIC) is missing entirely. You have to connect a wire to any unused GPIO pin (I selected pin 33 for consistency with v21new).

For v21new, all radio pins are connected to the ESP32 and there are the appropriate defines in `pins_arduino.h`
````
// SPI LoRa Radio
#define LORA_SCK    5   // GPIO5 - SX1276 SCK
#define LORA_MISO   19  // GPIO19 - SX1276 MISO
#define LORA_MOSI   27  // GPIO27 - SX1276 MOSI
#define LORA_CS     18  // GPIO18 - SX1276 CS
#define LORA_RST    23  // GPIO23 - SX1276 RST
#define LORA_IRQ    26  // GPIO26 - SX1276 IRQ (interrupt request)
#define LORA_D1     33  // GPIO33 - SX1276 IO1 (for LMIC Arduino library)
#define LORA_D2     32 // GPIO32 - SX1276 IO2
```

matthias-bs, 2023-02-13:
According to `https://github.com/matthias-bs/TTGO-ESP32-Bresser-with-display/blob/main/BresserWeatherSensorTTN2.ino` (I did not change your pin numbers) you used the following:
```
#if defined(ARDUINO_TTGO_LoRa32_V1)
// https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v1/pins_arduino.h
// http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1130&FId=t3:50003:3
// https://github.com/Xinyuan-LilyGo/TTGO-LoRa-Series
// https://github.com/LilyGO/TTGO-LORA32/blob/master/schematic1in6.pdf
#define PIN_LMIC_NSS LORA_CS
#define PIN_LMIC_RST LORA_RST
#define PIN_LMIC_DIO0 LORA_IRQ
#define PIN_LMIC_DIO1 33
#define PIN_LMIC_DIO2 cMyLoRaWAN::lmic_pinmap::LMIC_UNUSED_PIN
#else
```
(Provided `ARDUINO_TTGO_LoRa32_V1` was really set.)

Maybe I made some mistake or maybe you have an inconsistent mix of old and new code...

In pins_arduino.h, OLED_SCL, OLED_SDA and OLED_RST are already defined (so please do not re-define it in your sketch). According to this, your board should be V2 or V21new.

Again from `variants/ttgo-lora32-v21new/pins_arduino.h`:
```
// I2C OLED Display works with SSD1306 driver
#define OLED_SDA    21
#define OLED_SCL    22
#define OLED_RST    16
```

matthias-bs, 2023-02-13:
Again from our discussion three weeks ago: https://github.com/matthias-bs/BresserWeatherSensorTTN/issues/38#issuecomment-1398869861
According to this, you have a V2 board (which needs the extra wire) and you decided to connect it to pin 12, so you have to change this accordingly:
```
#define PIN_LMIC_DIO1 33
```
I'll call it a day now, otherwise I'll get a headache...

mczakk, 2023-02-13:
> I'll call it a day now, otherwise I'll get a headache...
Ill have a look at your comments in the morning. 
Really do appreciate your time in trying to get this sorted for me. 😁👍

mczakk, 2023-02-14:
I think I might have found the issue:
I told you that I'd moved to my linux laptop to use the arduino ide- 
when i looked at the board managers, it seems that the linux ide has a later version of 'esp32 by espressif systems' installed (v2.0.6)
the screenshots below show the difference in the board managers:
linux IDE
[image]
windows ide:
[image]
as you can see, the boards available in the windows (v 1.0.6) version of esp32 are different, and there is no option to select 'board revision'
I tried uninstalling the v1.0.6 version in the windows ide, and when i entered the same url in the 'additional board managers' section in preferences, i still got v 1.0.6, with no option to install v2.0.6.
installing the sketch works perfectly using the windows ide.

matthias-bs, 2023-02-14:
I think the board selection can be changed by editing the "Board Manager URLs" in the Preferences:
[image]

I remember I had a similar problem a while ago.

mczakk, 2023-02-14:
Yes, that is what is confusing me, I entered the same url  (https://dl.espressif.com/dl/package_esp32_index.json) into the board manager preferences on both my windows and linux machines. the windows machine gives me v1.0.6 (and the sketch compiles and uploads perfectly with the old code) and the linux machine gives me v2.0.6 and doesnt compile properly (giving the errors discussed above)
I will try using the same url as in your screenshot and see what happens

mczakk, 2023-02-14:
ok, changed board manager preferences to that above, installed additional libraries - theengsdecoder and nimble.
downloaded latest versions of BresserWeatherSensorTTN, WeatherSensorReciever and TTGO-ESP32-bresser-with-display to windows machine, board set to TTGO Lora32-OLED revision V2
waiting on compilation..... and it fails to work correctly, compiles and uploads ok, although i do get this message:
```
In file included from C:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorTTN2\BresserWeatherSensorTTN2.ino:173:
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:92:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver")
                                                                                 ^
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:93:66: note: #pragma message: LoRa DIO1 must be wired to GPIO33 manually!
     #pragma message("LoRa DIO1 must be wired to GPIO33 manually!")
                                                                  ^
In file included from C:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorTTN2\BresserWeatherSensorTTN2.ino:173:
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->12, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
c:\Users\mtc20tfq\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
c:\Users\mtc20tfq\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
                                                                        ^
In file included from c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensor.cpp:62:
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:92:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver")
                                                                                 ^
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:93:66: note: #pragma message: LoRa DIO1 must be wired to GPIO33 manually!
     #pragma message("LoRa DIO1 must be wired to GPIO33 manually!")
                                                                  ^
In file included from c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensor.cpp:62:
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->12, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->33
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
```
and corresponding serial monitor output, with no connection to TTN:
```
17:21:38.127 -> load:0x3fff0030,len:1184
17:21:38.127 -> load:0x40078000,len:13132
17:21:38.127 -> load:0x40080400,len:3036
17:21:38.127 -> entry 0x400805e4
17:22:02.112 -> ets Jun  8 2016 00:22:57
17:22:02.112 -> 
17:22:02.112 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:22:02.112 -> configsip: 188777542, SPIWP:0xee
17:22:02.112 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:22:02.112 -> mode:DIO, clock div:1
17:22:02.112 -> load:0x3fff0030,len:1184
17:22:02.112 -> load:0x40078000,len:13132
17:22:02.112 -> load:0x40080400,len:3036
17:22:02.112 -> entry 0x400805e4
17:22:26.108 -> ets Jun  8 2016 00:22:57
17:22:26.108 -> 
17:22:26.108 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:22:26.108 -> configsip: 188777542, SPIWP:0xee
17:22:26.108 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:22:26.108 -> mode:DIO, clock div:1
17:22:26.108 -> load:0x3fff0030,len:1184
17:22:26.108 -> load:0x40078000,len:13132
17:22:26.108 -> load:0x40080400,len:3036
17:22:26.108 -> entry 0x400805e4
````

matthias-bs, 2023-02-14:
The messages are just for information which pins and which transceiver will be used (based on the board settings).

The question is: which pin actually worked for you or to which ESP32 GPIO pin did you connect the DIO1 wire?

If I got it right from the conversation, you used GPIO12 (not GPIO33). If this is true, you have to change the define for `PIN_LMIC_DIO1` accordingly.

matthias-bs, 2023-02-14:
BTW: You are switching between two computers and updating the Arduino development environments... Did you remember to apply the configurations and fixes for the LMIC libraries?

mczakk, 2023-02-14:
Ah, I got confused because some of the images of pin maps I've seen have pins labelled as different numbers!
I will change PIN_LMIC_DI01 TO 12, just in ..TTN2.ino yes?
Will see if it compiles properly now!

mczakk, 2023-02-14:
nope, still didn't work, was wondering if i needed to change this bit? (lines 187 to 205 in BresserWeatherSensorTTN2.ino
```
#if defined(ARDUINO_TTGO_LoRa32_V1)
// https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v1/pins_arduino.h
// http://www.lilygo.cn/prod_view.aspx?TypeId=50003&Id=1130&FId=t3:50003:3
// https://github.com/Xinyuan-LilyGo/TTGO-LoRa-Series
// https://github.com/LilyGO/TTGO-LORA32/blob/master/schematic1in6.pdf
#define PIN_LMIC_NSS LORA_CS
#define PIN_LMIC_RST LORA_RST
#define PIN_LMIC_DIO0 LORA_IRQ
#define PIN_LMIC_DIO1 12
#define PIN_LMIC_DIO2 cMyLoRaWAN::lmic_pinmap::LMIC_UNUSED_PIN
#else
// LoRaWAN_Node board
// https://github.com/matthias-bs/LoRaWAN_Node
#define PIN_LMIC_NSS 14
#define PIN_LMIC_RST 12
#define PIN_LMIC_DIO0 4
#define PIN_LMIC_DIO1 16
#define PIN_LMIC_DIO2 17
#endif
```
also changed 33 to 12 in WeatherSensorRecievierCFG.h : 
```
#elif defined(ARDUINO_TTGO_LoRa32_V1) || defined(ARDUINO_TTGO_LoRa32_V2)
    // Use pinning for LILIGO TTGO LoRa32-OLED
    #define PIN_RECEIVER_CS   LORA_CS
    
    // CC1101: GDO0 / RFM95W/SX127x: G0
    #define PIN_RECEIVER_IRQ  LORA_IRQ
    
    // CC1101: GDO2 / RFM95W/SX127x: G1
    // n.c. on v1/v2?, LORA_D1 on v21
    #define PIN_RECEIVER_GPIO 12
    
    // RFM95W/SX127x - GPIOxx / CC1101 - RADIOLIB_NC
    #define PIN_RECEIVER_RST  LORA_RST
```
still get 

```
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:313:170: note: #pragma message: Pin config: RST->12, CS->18, GD0/G0/IRQ->26, GDO2/G1/GPIO->12
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
c:\Users\mtc20tfq\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
c:\Users\mtc20tfq\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
                                                                        ^
In file included from c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensor.cpp:62:
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:92:81: note: #pragma message: ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver
     #pragma message("ARDUINO_TTGO_LoRa32_V2 defined; using on-board transceiver")
                                                                                 ^
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:93:66: note: #pragma message: LoRa DIO1 must be wired to GPIO33 manually!
     #pragma message("LoRa DIO1 must be wired to GPIO33 manually!")
                                                                  ^
In file included from c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensor.cpp:62:
c:\Users\mtc20tfq\Arduino\libraries\BresserWeatherSensorReceiver\src\WeatherSensorCfg.h:312:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
```
on compiling

mczakk, 2023-02-14:
The pin config bit does say RST - >12  and GDO2/G1/GPIO->12, so I guess that's where the problem lies!

matthias-bs, 2023-02-15:
Well, the question is not if it compiles properly, but if the pins matching your board (and your extra wiring) are selected!
The compiler can't know this! The pragma messages are just for information how the defines are finally translated to the GPIO numbers.

To go one step back, please check which boards (are all your boards of the same version?) you have (e.g. by comparing it with https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973) and check which pinout your board has. If your board is this weird V2 version, check if you have extra wire for DIO1 and to which GPIO it is connected. Then make sure this GPIO is selected in the defines you are going to uses. Then compile and compare the pragma messages with the GPIO numbers you intended to be used.

If all this fails, please go back to the versions (according to the date) and IDE settings which worked for you previously (and then check what has been changed since then).

matthias-bs, 2023-02-15:
> The pin config bit does say RST - >12 and GDO2/G1/GPIO->12, so I guess that's where the problem lies!

What the...

in
`https://github.com/espressif/arduino-esp32/blob/master/variants/ttgo-lora32-v2/pins_arduino.h`:
```
#define LORA_RST    12  // GPIO14 - SX1276 RST
```
So the actual define and the comment are different!!!

According to https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973, the radio's reset pin is actually connected to the HW reset and therefore cannot be controlled by SW. Therefore they propose to assign `cMyLoRaWAN::lmic_pinmap::LMIC_UNUSED_PIN` to it! What a ****!

mczakk, 2023-02-15:
> To go one step back, please check which boards (are all your boards of the same version?) you have (e.g. by comparing it with https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic-part-2/11973) and check which pinout your board has. If your board is this weird V2 version, check if you have the extra wire for DIO1 and to which GPIO it is connected. Then make sure this GPIO is selected in the defines you are going to uses. Then compile and compare the pragma messages with the GPIO numbers you intended to be used.

i checked that thread and the board i have isn't shown on there - these are the boards i have (i have two, and they are slightly different, one is marked T3V1.6, (this one is working ok out in the field), the one i am currently trying to get to work is marked T3_v1.6.1)):
[image]
[image]

> 
> If all this fails, please go back to the versions (according to the date) and IDE settings which worked for you previously (and then check what has been changed since then).

The main thing that has changed is that since the update of the esp32 board manager from 1.0.6 to 2.0.6 all the errors have occurred.
as i stated in my post above, version 1.0.6 only has one option for TTGO LoRa32 OLED V1, whereas v2.0.6 has TTGO LoRa32 OLED with three different options in the 'board revisions bit of the 'Tools' dropdown (which isn't present in V1.0.6)

OK, not sure i understand this, but just tried compiling with board revision set to TTGO LoRa v1 (NoTFcard) and it seems to work ok, reporting and displaying ok, but the message sent to ttn is different, so the old payload formatter is faioling with a mask length error!

matthias-bs, 2023-02-15:
Which errors have occurred?

T3_V1.6.1 should be compatible to v21new (http://www.lilygo.cn/prod_view.aspx?TypeId=50060&Id=1271&FId=t3:50060:3)

The payload length should only depend on the selected options in BresserWeatherSensorTTNCfg.h.

Maybe the change comes from the way I'm dealing with `PIN_ADC3_IN`. I'm only defining this for my own LORAWAN_NODE board, because I currently do not have sufficient information on how to use this in a useful way for other boards.

So, for your purpose, if you have updated `BresserWeatherSensorTTNCfg.h` from GitHub, just enable it for your board as with the previous version and the payload length should be the same as before (even if the reported voltage is not useful).

matthias-bs, 2023-02-15:
https://github.com/lnlp/pinout-diagrams/blob/main/LoRa%20development%20boards/TTGO%20LoRa32%20V2.1.6%20Pinout%20(LLP).pdf

mczakk, 2023-02-15:
looks like the board I have in the feild is a v1.6, and the one on my desk is a v1.6.1
Think i've solved the payload length issue, set #define SENSOR_ID to on, will see what happens, at the moment its taking 10+ minutes to compile for some rreason.....

matthias-bs, 2023-02-15:
Ah, ok. If you are using `decoder_basic.js`, you are expecting the ID (I did not remember that), but you are also expecting supply_v **and** battery_v (i.e. PIN_ADC3_IN has to be defined).

mczakk, 2023-02-15:
sorted it out now, PIN_ADC3_IN was not defined.
works fine (but compiles very slowly) on windows now, 
still having a problem compiling it from linux - something to do with esp32 Preferences.cpp

mczakk, 2023-02-17:
All seems to be working ok now, though for some reason it refuses to compile using Arduino ide 2.0 on my Ubuntu 22.04 laptop (works fine using IDE 1.8 on another Ubuntu 22.04 one though!)
one thing that would be useful from a data representation point of view - the part in BresserWeatherSensorTTN.ino (lines1395-1423) where it checks the weather data and fills it with suspicious dummy values - is it possible to NOT report to TTN if those values are present? It makes a mess of my graphs!!
Thanks again for all your help with this :)

matthias-bs, 2023-02-17:
Would you like to investigate why it fails with IDE 2.0?

For the second issue: please check the define WEATHERSENSOR_DATA_REQUIRED. If this define is set and weather data is incomplete/not available, the sketch enters sleep mode without sending data.

Another solution would be to process the data only if the valid-flag was set.

mczakk, 2023-02-17:
> Would you like to investigate why it fails with IDE 2.0?
I'll try, the ESP decoder you mentioned doesnt work with ide 2.0, so i cant decoded the stack trace or backtrace, I'll post them on here, but i need to have the board running to collect data, 
> For the second issue: please check the define WEATHERSENSOR_DATA_REQUIRED. If this define is set and weather data is incomplete/not available, the sketch enters sleep mode without sending data.
trying with this.....
> Another solution would be to process the data only if the valid-flag was set.

mczakk, 2023-02-17:
serial monitor output after compiling on ubuntu 22.04 and arduino IDE 2.0.3
```
14:26:43.959 -> ELF file SHA256: 8889a6ae06c0051b
14:26:43.959 -> 
14:26:44.086 -> Rebooting...
14:26:44.118 -> ets Jun  8 2016 00:22:57
14:26:44.118 -> 
14:26:44.118 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:26:44.118 -> configsip: 188777542, SPIWP:0xee
14:26:44.118 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:26:44.118 -> mode:DIO, clock div:1
14:26:44.118 -> load:0x3fff0030,len:1184
14:26:44.118 -> load:0x40078000,len:13132
14:26:44.118 -> load:0x40080400,len:3036
14:26:44.118 -> entry 0x400805e4
14:26:44.408 -> [    19][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
14:26:44.921 -> [   532][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: ws_timeout NOT_FOUND
14:26:44.921 -> [   532][D][BresserWeatherSensorTTN2.ino:598] setup(): Preferences: weathersensor_timeout: 180 s
14:26:44.921 -> [   537][V][Preferences.cpp:376] getUShort(): nvs_get_u16 fail: sleep_interval NOT_FOUND
14:26:44.953 -> [   545][D][BresserWeatherSensorTTN2.ino:600] setup(): Preferences: sleep_interval:        360 s
14:26:44.953 -> [   553][V][Preferences.cpp:376] getUShort(): nvs_get_u16 fail: sleep_interval_long NOT_FOUND
14:26:44.953 -> [   562][D][BresserWeatherSensorTTN2.ino:602] setup(): Preferences: sleep_interval_long:   900 s
14:26:44.953 -> [   570][D][BresserWeatherSensorTTN2.ino:607] setup(): 
14:26:44.953 -> [   576][D][BresserWeatherSensorTTN2.ino:992] printDateTime(): 00:01:03 01-01-70
14:26:44.985 -> [   582][D][BresserWeatherSensorTTN2.ino:615] setup(): RTC sync required
14:26:44.985 -> [   589][D][BresserWeatherSensorTTN2.ino:621] setup(): myEventlog.setup() - done
14:26:44.985 -> [   598][D][BresserWeatherSensorTTN2.ino:1193] getVoltage(): Voltage = 4108mV
14:26:44.985 -> [   603][D][WeatherSensor.cpp:82] begin(): [SX1276] Initializing ... 
14:26:45.020 -> [   620][D][WeatherSensor.cpp:95] begin(): success!
14:26:45.020 -> [   621][D][WeatherSensor.cpp:130] begin(): [SX1276] Setup complete - awaiting incoming messages...
14:26:55.665 -> [ 11268][V][WeatherSensor.cpp:218] getMessage(): [SX1276] Data: D4 7A 1B 38 05 35 05 18 FD CE FD 21 68 12 36 88 FF F0 60 AA 00 00 00 00 00 00 00 
14:26:55.665 -> [ 11270][D][WeatherSensor.cpp:220] getMessage(): [SX1276] R [D4] RSSI: -80.5
14:26:55.665 -> [ 11277][V][WeatherSensor.cpp:285] findSlot(): find_slot(): ID=38053505
14:26:55.665 -> [ 11283][V][WeatherSensor.cpp:339] findSlot(): find_slot(): Storing into slot #0
14:26:55.665 -> [ 11290][D][WeatherSensor.cpp:756] decodeBresser6In1Payload(): Msg flags - Temp: 1  Hum: 1  UV: 1  Wind: 1  Rain: 0  Moist: 0
14:27:07.666 -> [ 23275][V][WeatherSensor.cpp:218] getMessage(): [SX1276] Data: D4 4C 34 38 05 35 05 18 FE CC FE 21 68 FF 6D DB FF 01 D8 AA 00 00 00 00 00 00 00 
14:27:07.666 -> [ 23276][D][WeatherSensor.cpp:220] getMessage(): [SX1276] R [D4] RSSI: -80.0
14:27:07.666 -> [ 23283][V][WeatherSensor.cpp:285] findSlot(): find_slot(): ID=38053505
14:27:07.666 -> [ 23289][V][WeatherSensor.cpp:333] findSlot(): find_slot(): Updating slot #0
14:27:07.666 -> [ 23296][D][WeatherSensor.cpp:756] decodeBresser6In1Payload(): Msg flags - Temp: 0  Hum: 0  UV: 0  Wind: 1  Rain: 1  Moist: 0
14:27:07.697 -> [ 23307][D][BresserWeatherSensorTTN2.ino:1124] setup(): Receiving Weather Sensor Data o.k.
14:27:07.697 -> [ 23315][D][BresserWeatherSensorTTN2.ino:625] setup(): mySensor.setup() - done
14:27:07.729 -> [ 23338][D][BresserWeatherSensorTTN2.ino:933] NetGetSessionState(): failed
14:27:07.729 -> [ 23338][D][BresserWeatherSensorTTN2.ino:629] setup(): myLoRaWAN.setup() - done
14:27:07.729 -> [ 23341][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
14:27:07.729 -> Display should work
14:27:07.729 -> [ 23355][D][BresserWeatherSensorTTN2.ino:1193] getVoltage(): Voltage = 4294mV
14:27:07.729 -> [ 23359][D][BresserWeatherSensorTTN2.ino:1209] getVoltage(): Voltage = 321mV
14:27:07.761 -> [ 23364][D][BresserWeatherSensorTTN2.ino:1315] doUplink(): --- Uplink Data ---
14:27:07.761 -> [ 23371][D][BresserWeatherSensorTTN2.ino:1319] doUplink(): Air Temperature:     12.3 °C
14:27:07.761 -> [ 23379][D][BresserWeatherSensorTTN2.ino:1320] doUplink(): Humidity:            88   %
14:27:07.793 -> [ 23387][D][BresserWeatherSensorTTN2.ino:1321] doUplink(): Rain Gauge:         922.4 mm
14:27:07.793 -> [ 23394][D][BresserWeatherSensorTTN2.ino:1322] doUplink(): Wind Speed (avg.):    1.3 m/s
14:27:07.793 -> [ 23402][D][BresserWeatherSensorTTN2.ino:1323] doUplink(): Wind Speed (max.):    1.3 m/s
14:27:07.793 -> [ 23410][D][BresserWeatherSensorTTN2.ino:1324] doUplink(): Wind Direction:     216.0 °
14:27:07.825 -> [ 23418][D][BresserWeatherSensorTTN2.ino:1350] doUplink(): Supply  Voltage:   4294   mV
14:27:07.825 -> [ 23425][D][BresserWeatherSensorTTN2.ino:1353] doUplink(): Battery Voltage:    321   mV
14:27:07.825 -> [ 23433][D][BresserWeatherSensorTTN2.ino:1369] doUplink(): 
14:27:07.825 -> [ 23438][I][BresserWeatherSensorTTN2.ino:1378] doUplink(): sensor_id: 939865349
14:27:07.825 -> [ 23445][D][BresserWeatherSensorTTN2.ino:1459] doUplink(): Rain past 60min:      0.0 mm
14:27:07.857 -> [ 23453][D][BresserWeatherSensorTTN2.ino:1460] doUplink(): Rain curr. day:       0.0 mm
14:27:07.858 -> [ 23461][D][BresserWeatherSensorTTN2.ino:1461] doUplink(): Rain curr. week:      0.0 mm
[... cut ...]
14:27:07.921 -> EXCVADDR: 0x00847658  LBEG    : 0x400871b4  LEND    : 0x400871bf  LCOUNT  : 0x00000000  
14:27:07.921 -> 
14:27:07.921 -> 
14:27:07.921 -> Backtrace: 0x400d4fca:0x3ffc9860 0x400d5071:0x3ffc9890 0x400d4b5b:0x3ffc9990 0x400d4bf4:0x3ffc99b0 0x400d4c34:0x3ffc99d0 0x400d5b51:0x3ffc99f0 0x400d6fb9:0x3ffc9a10 0x400d74f6:0x3ffc9a30 0x400d8361:0x3ffc9a50 0x400d4eab:0x3ffc9a70 0x400d47e9:0x3ffc9a90 0x400e0efd:0x3ffc9ab0
14:27:07.953 -> 
14:27:07.953 -> 
14:27:07.953 -> 
14:27:07.953 -> 
14:27:07.953 -> ELF file SHA256: 8889a6ae06c0051b
14:27:07.953 -> 
14:27:08.113 -> Rebooting...
```
```

matthias-bs, 2023-02-17:
The verbose output from Preferences.cpp is not an error: If the settings are changed via LoRaWAN downlink, they are stored via the Preferences library in the flash. If settings have not been changed yet, their default values are used.

mczakk, 2023-02-17:
whats the 
```Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
14:27:07.889 -> 
14:27:07.889 -> Core  1 register dump:
14:27:07.889 -> PC      : 0x400d4fcd  PS      : 0x00060630  A0      : 0x800d5074  A1      : 0x3ffc9860  
14:27:07.889 -> A2      : 0x3ffc98ec  A3      : 0x00847628  A4      : 0x00000084  A5      : 0x3ffc5f04  
14:27:07.889 -> A6      : 0x00008476  A7      : 0x3ffc98cc  A8      : 0x3ffc5f04  A9      : 0x33be27a0  
14:27:07.921 -> A10     : 0x3ffc98cc  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc5f04  
14:27:07.921 -> A14     : 0x3ffc991c  A15     : 0x63ffffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
14:27:07.921 -> EXCVADDR: 0x00847658  LBEG    : 0x400871b4  LEND    : 0x400871bf  LCOUNT  : 0x00000000  
14:27:07.921 -> 
14:27:07.921 -> 
14:27:07.921 -> Backtrace: 0x400d4fca:0x3ffc9860 0x400d5071:0x3ffc9890 0x400d4b5b:0x3ffc9990 0x400d4bf4:0x3ffc99b0 0x400d4c34:0x3ffc99d0 0x400d5b51:0x3ffc99f0 0x400d6fb9:0x3ffc9a10 0x400d74f6:0x3ffc9a30 0x400d8361:0x3ffc9a50 0x400d4eab:0x3ffc9a70 0x400d47e9:0x3ffc9a90 0x400e0efd:0x3ffc9ab0
```
bit about then? :)

matthias-bs, 2023-02-17:
You could try this ESP Stack Trace Decoder: https://github.com/littleyoda/EspStackTraceDecoder