matthias-bs/BresserWeatherSensorTTN #49: Request for testing Heltec Wireless Stick configuration

matthias-bs, 2023-03-04:
Please test the [Heltec Wireless Stick](https://heltec.org/project/wireless-stick/) configuration and provide your feedback!

Deep-sleep current consumption would be interesting, too!

matthias-bs, 2023-03-04:
@jorgenavarroortiz It would be great if you could spend a little time on this!

matthias-bs, 2023-03-04:
Please refer to https://github.com/matthias-bs/BresserWeatherSensorTTN#software-build-setup and select the following board:
[image]

jorgenavarroortiz, 2023-03-05:
Sure! Just give me a few days and I will try to connect this board to TTN :-)

jorgenavarroortiz, 2023-03-08:
Hi! The pinout is ok since I am receiving data on TTN. However, I think I have some problem with the decoder (ttn_decoder_fp.js) since the data do not show proper values. If you have some simple steps that I can follow to test this, it would be great. On BresserWeatherSensorTTNCfg.h I have only enabled SENSORID_EN and RAINDATA_EN, and I have updated the javascript formatter on TTN. However, results are strange... I will try to make it work in the next days.

matthias-bs, 2023-03-09:
Hi,
Thank you very much for testing!

In https://github.com/matthias-bs/BresserWeatherSensorTTN/blob/main/decoder_basic.js there is a more detailed explanation how to set the decoder based on the Arduino sketch configuration.