matthias-bs/BresserWeatherSensorTTN #73: Firebeetle LoRa Cover

KennyM64, 2023-10-03:
Firebeetle LoRa Cover on Firebeetle ESP32 with nss->27/D4, rst->25/D2, DIO0->26/D3, DIO1->9/D5, DIO2->LMIC_UNUSED_PIN

When compiled and uploaded the board boot/rst loops.
`rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13400
load:0x40080400,len:3672
entry 0x400805f8
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1412
load:0x40078000,len:13400
load:0x40080400,len:3672
entry 0x400805f8
ets Jun  8 2016 00:22:57`

Full Compile Output
`G:\My Drive\Arduino\Smart Cities\BresserWeatherSensorTTN-main\BresserWeatherSensorTTN\BresserWeatherSensorTTN.ino:166:6: warning: #warning "LMIC_ENABLE_DeviceTimeReq is not set - will not be able to retrieve network time!" [-Wcpp]
     #warning "LMIC_ENABLE_DeviceTimeReq is not set - will not be able to retrieve network time!"
      ^~~~~~~
G:\My Drive\Arduino\Smart Cities\BresserWeatherSensorTTN-main\BresserWeatherSensorTTN\BresserWeatherSensorTTN.ino:691:3: warning: #warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!" [-Wcpp]
 # warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
   ^~~~~~~
In file included from c:\Users\cow5038\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/Lightning.h:63,
                 from G:\My Drive\Arduino\Smart Cities\BresserWeatherSensorTTN-main\BresserWeatherSensorTTN\BresserWeatherSensorTTN.ino:160:
c:\Users\cow5038\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:164:140: note: #pragma message: LORAWAN_NODE defined; assuming this is the LoRaWAN_Node board (DFRobot Firebeetle32 + Adafruit RFM95W LoRa Radio)
         #pragma message("LORAWAN_NODE defined; assuming this is the LoRaWAN_Node board (DFRobot Firebeetle32 + Adafruit RFM95W LoRa Radio)")
                                                                                                                                            ^
c:\Users\cow5038\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:491:48: note: #pragma message: Receiver chip: [SX1276]
 #pragma message("Receiver chip: " RECEIVER_CHIP)
                                                ^
c:\Users\cow5038\Documents\Arduino\libraries\BresserWeatherSensorReceiver\src/WeatherSensorCfg.h:492:170: note: #pragma message: Pin config: RST->12, CS->14, GD0/G0/IRQ->4, GDO2/G1/GPIO->16
 #pragma message("Pin config: RST->" STR(PIN_RECEIVER_RST) ", CS->" STR(PIN_RECEIVER_CS) ", GD0/G0/IRQ->" STR(PIN_RECEIVER_IRQ) ", GDO2/G1/GPIO->" STR(PIN_RECEIVER_GPIO) )
                                                                                                                                                                          ^
G:\My Drive\Arduino\Smart Cities\BresserWeatherSensorTTN-main\BresserWeatherSensorTTN\BresserWeatherSensorTTN.ino:271:119: note: #pragma message: FIREBEETLE_COVER_LORA defined; assuming FireBeetle ESP32 with FireBeetle Cover LoRa will be used
     #pragma message("FIREBEETLE_COVER_LORA defined; assuming FireBeetle ESP32 with FireBeetle Cover LoRa will be used")
                                                                                                                       ^
G:\My Drive\Arduino\Smart Cities\BresserWeatherSensorTTN-main\BresserWeatherSensorTTN\BresserWeatherSensorTTN.ino:272:85: note: #pragma message: Required wiring: D2 to RESET, D3 to DIO0, D4 to CS, D5 to DIO1
     #pragma message("Required wiring: D2 to RESET, D3 to DIO0, D4 to CS, D5 to DIO1")
                                                                                     ^
c:\Users\cow5038\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
c:\Users\cow5038\Documents\Arduino\libraries\MCCI_LoRaWAN_LMIC_library\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")`

KennyM64, 2023-10-09:
#define FIREBEETLE_ESP32_COVER_LORA was defined and pragma message shows the correct board being selected but still doing a boot loop :-(

matthias-bs, 2023-10-09:
I might be able to help if you post more of your code, Arduino IDE settings, compiler output and log messages.

matthias-bs, 2023-10-09:
Could you please make the following changes in `BresserWeatherSensorTTNCfg.h`:
```
//--- Select Board ---
#if !defined(ARDUINO_TTGO_LoRa32_V1)     && !defined(ARDUINO_TTGO_LoRa32_V2) && \
    !defined(ARDUINO_TTGO_LoRa32_v21new) && !defined(ARDUINO_ADAFRUIT_FEATHER_ESP32S2) && \
    !defined(ARDUINO_FEATHER_ESP32)      && !defined(ARDUINO_ADAFRUIT_FEATHER_RP2040)
    // Use pinning for LoRaWAN Node 
    //#define LORAWAN_NODE

    // Use pinning for Firebeetle Cover LoRa
    #define FIREBEETLE_COVER_LORA
#endif
```

It seems I missed this when backporting the Firebeetle configuration from my growatt2lorawan project.

matthias-bs, 2023-10-09:
You also have to set `#define BATTERY_LOW 0` until you make the required solder connections.

matthias-bs, 2023-10-09:
Both changes above are provided in [Release v0.11.1](https://github.com/matthias-bs/BresserWeatherSensorTTN/releases/tag/v0.11.1), which should be available within the next hour.
This time I made a successful test run. :sweat_smile:

KennyM64, 2023-10-09:
Thanks, Matthias that seems to have done the trick. Strangely I had to use a different computer as my main computer still boot/loops but the other doesn't. I'd say I have an issue with the Arduino install or library setup on the boot/loop computer so I may blow it away and start a fresh.
Thanks again for your hard work. The program is great.

matthias-bs, 2023-10-10:
Sometimes outdated compile results are still used. You can try to find out where Arduino stores temporary data and delete it. (I don't use Windows for this, therefore I can't tell.)

closed 2023-10-10 by matthias-bs